labscript-suite/blacs #71: Subprocesses fail to launch

rpanderson, 2020-05-30:
Tabs fail to initialise, cannot recompile connection table, etc. owing to the following, which is being raised in a new shell for every subprocess:
```python-traceback
Traceback (most recent call last):
  File "c:\users\rpanderson\labscript-suite-testing\.venv\lib\site-packages\zprocess\process_tree.py", line 1478, in connect_to_parent
    parentinfo = json.loads(os.environ['ZPROCESS_PARENTINFO'])
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.752.0_x64__qbz5n2kfra8p0\lib\os.py", line 675, in __getitem__
    raise KeyError(key) from None
KeyError: 'ZPROCESS_PARENTINFO'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.752.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.752.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "c:\users\rpanderson\labscript-suite-testing\.venv\lib\site-packages\zprocess\process_class_wrapper.py", line 86, in <module>
    _setup()
  File "c:\users\rpanderson\labscript-suite-testing\.venv\lib\site-packages\zprocess\process_class_wrapper.py", line 34, in _setup
    process_tree = ProcessTree.connect_to_parent()
  File "c:\users\rpanderson\labscript-suite-testing\.venv\lib\site-packages\zprocess\process_tree.py", line 1483, in connect_to_parent
    raise RuntimeError(msg)
RuntimeError: No ZPROCESS_PARENTINFO environment variable found
```

* I have zmq encryption enabled and running with the correct PSK

Relates to chrisjbillington/zprocess@b4b1bf83.

Output at launch:
```
2020-05-30 17:47:05,764 INFO BLACS: Python version 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 23:03:10) [MSC v.1916 64 bit (AMD64)]
2020-05-30 17:47:05,764 INFO BLACS: Platform: win32
2020-05-30 17:47:05,764 INFO BLACS: windows version: sys.getwindowsversion(major=10, minor=0, build=18362, platform=2, service_pack='')
2020-05-30 17:47:05,765 INFO BLACS: PyZMQ version: 19.0.1
2020-05-30 17:47:05,766 INFO BLACS: ZMQ version: 4.3.2
2020-05-30 17:47:05,767 INFO BLACS: h5py version: Summary of the h5py configuration
---------------------------------

h5py    2.10.0
HDF5    1.10.5
Python  3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 23:03:10) [MSC v.1916 64 bit (AMD64)]
sys.platform    win32
sys.maxsize     9223372036854775807
numpy   1.18.4

2020-05-30 17:47:05,768 INFO BLACS: Qt enviroment: PyQt5
2020-05-30 17:47:05,768 INFO BLACS: PySide/PyQt version: 5.14.2
2020-05-30 17:47:05,769 INFO BLACS: Qt version: 5.14.2
2020-05-30 17:47:05,769 INFO BLACS: qtutils version: 2.3.2
2020-05-30 17:47:05,770 INFO BLACS: zprocess version: 2.21.0.dev1
2020-05-30 17:47:05,771 INFO BLACS: labscript_utils version: 3.1.0.dev3
2020-05-30 17:47:05,771 INFO BLACS: BLACS version: 3.1.0.dev3
```

rpanderson, 2020-05-30:
This is with (Windows) Store-bought Python. On a fresh conda install I don't observe this issue. There. I have:
```
2020-05-30 19:46:04,157 INFO BLACS: Python version 3.8.3 (default, May 19 2020, 06:50:17) [MSC v.1916 64 bit (AMD64)]
2020-05-30 19:46:04,157 INFO BLACS: Platform: win32
2020-05-30 19:46:04,158 INFO BLACS: windows version: sys.getwindowsversion(major=10, minor=0, build=18362, platform=2, service_pack='')
2020-05-30 19:46:04,159 INFO BLACS: PyZMQ version: 19.0.1
2020-05-30 19:46:04,159 INFO BLACS: ZMQ version: 4.3.2
2020-05-30 19:46:04,160 INFO BLACS: h5py version: Summary of the h5py configuration
---------------------------------

h5py    2.10.0
HDF5    1.10.5
Python  3.8.3 (default, May 19 2020, 06:50:17) [MSC v.1916 64 bit (AMD64)]
sys.platform    win32
sys.maxsize     9223372036854775807
numpy   1.18.4

2020-05-30 19:46:04,161 INFO BLACS: Qt enviroment: PyQt5
2020-05-30 19:46:04,161 INFO BLACS: PySide/PyQt version: 5.12.3
2020-05-30 19:46:04,161 INFO BLACS: Qt version: 5.12.5
2020-05-30 19:46:04,162 INFO BLACS: qtutils version: 2.4.0.dev2
2020-05-30 19:46:04,162 INFO BLACS: zprocess version: 2.19.0.dev1
2020-05-30 19:46:04,163 INFO BLACS: labscript_utils version: 3.1.0.dev3
2020-05-30 19:46:04,163 INFO BLACS: BLACS version: 3.1.0.dev3
```

rpanderson, 2020-05-30:
One potentially substantive difference is zprocess (2.19.0.dev1 vs 2.21.0.dev1 above). The former is not available on PyPI or TestPyPI, but installing zprocess 2.19.0 or 2.20.0 does not rectify the issue.

chrisjbillington, 2020-05-30:
Probably not related to chrisjbillington/zprocess@b4b1bf8 since the crash occurs one line earlier. Seems the environment variable isn't making it into the child processes in the first place (subsequently deleting it is to ensure it doesn't make it into grandchild processes etc).

I'll see if I can reproduce it. Perhaps there are some extra restrictions around environment variable inheritance for Microsoft Store apps.

chrisjbillington, 2020-05-30:
I can't reproduce this. Exact same versions of everything in Windows 10. Is this still specific to ConEmu, or are you seeing it from cmd.exe? I've tried both, each in 32 and 64 bit variants.

Are you able to run zprocess's hello-world?

```bash
$ python3 -m zprocess.examples.processclass_example
this is a running foo in process 7500
data is bar
foo, got a message: hello, foo!
parent, got a response: hello yourself!
```
I'll get together an example that will print debug info before trying to connect to the parent, and we can look at os.environ and see why it's missing ZPROCESS_PARENTINFO.

chrisjbillington, 2020-05-30:
Can't reproduce with or without a venv. I don't think the zprocess versions should make a difference - there are not really any code changes between these versions.

What working directory are you running from?

chrisjbillington, 2020-05-31:
This Python bug sounds potentially relevant:

https://bugs.python.org/issue38092

rpanderson, 2020-05-31:
> I can't reproduce this. Exact same versions of everything in Windows 10. Is this still specific to ConEmu, or are you seeing it from cmd.exe? I've tried both, each in 32 and 64 bit variants.

I'm running blacs from Windows Terminal, not ConEmu. However, the cmd.exe shells that launch and display the resulting traceback are in a ConEmu terminal, as that's the default terminal for cmd.exe on the host in question.

> Are you able to run zprocess's hello-world?

Yes.

> I'll get together an example that will print debug info before trying to connect to the parent, and we can look at os.environ and see why it's missing ZPROCESS_PARENTINFO.

Sounds good.

rpanderson, 2020-05-31:
> Perhaps there are some extra restrictions around environment variable inheritance for Microsoft Store apps.

I feel like we saw something very similar when debugging chrisjbillington/desktop-app recently.

chrisjbillington, 2020-05-31:
>>Are you able to run zprocess's hello-world?
> Yes.

That's promising!

> I feel like we saw something very similar when debugging chrisjbillington/desktop-app recently.

Yeah it was exactly the same

rpanderson, 2020-05-31:
Upon disabling ConEmu as the default terminal for console applications (namely cmd.exe) and removing `python.exe|python3.8.exe` from the list of hooked executables (_Settings >> Integration >> Default term_), I no longer observe this issue. Nor can I reproduce it upon reverting these settings to what they were when I did observe the issue :-(

In any case, I'm happy to put this down to hacky magic by ConEmu getting in the way, like that which we observed [a few weeks ago](https://github.com/labscript-suite/labscript-utils/pull/32#issuecomment-622665425) (not desktop-app).

Bizarrely, somewhere along the way _Settings >> Startup >> Command line_ in ConEmu got set to 
```
-new_console:c c:\users\rpanderson\labscript-suite-testing\.venv\scripts\python.exe -m zprocess.process_class_wrapper NovatechDDS9mWorker
```
... and not by me. This results in the very same traceback of the original post, but doesn't explain:
* How this setting got hijacked, or
* Why there was one of these for each tab.

In an asynchronous discussion during that time, you also referenced Maximus5/ConEmu#858 
 relating to environment variables in ConEmu.

This is worth keeping in mind, but seeing as I can no longer reproduce the issue I'm closing it and will keep it in mind if users observe this traceback (at which point we should ask if they're using ConEmu).

FWIW, although Windows Terminal is not the saviour the [promo video](https://www.youtube.com/watch?v=8gw0rXPMMPE) promised, it's a few orders of magnitude better than Command Prompt. Given this and the ConEmu funniness with Windows Store Python and virtual envs, I haven't used ConEmu in weeks.

closed 2020-05-31 by rpanderson