labscript-suite/blacs #74: Auto-restart fails in developer installation using Windows Store Python

rpanderson, 2020-06-12:
The 'Restart BLACS' functionality after recompiling the connection table doesn't work on a developer installation using Windows Store Python when launching blacs via either the `blacs` or `blacs-gui` commands from inside a virtual environment. This issue is not observed when launching blacs via the application shortcut for this installation; or for any launch mode of a regular install in:
  * another venv of a Windows Store Python distribution; or
  * a conda environment. 

Here's the output when using the `blacs` entry point in the developer venv:
```shell
2020-06-12 09:58:40,804 INFO BLACS: relaunching BLACS after quit
C:\Users\rpanderson\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\python.exe: can't open file 'c:\users\rpanderson\labscript-suite\.venv\Scripts\blacs-gui': [Errno 2] No such file or directory
```

chrisjbillington, 2020-06-12:
`'c:\users\rpanderson\labscript-suite\.venv\Scripts\blacs-gui': [Errno 2] No such file or directory`

Pretty weird. I'll see if I can reproduce.

chrisjbillington, 2020-06-12:
Not able to reproduce the issue in a fresh Windows-store venv in the same location on disk. Could you:

a) run `where blacs-gui` and `where blacs` with the venv activated and report the results and
b) confirm if `c:\users\rpanderson\labscript-suite\.venv\Scripts\blacs-gui` actually exists?

I'm at a bit of a loss at the moment, none of the explanations I've thought of so far are consistent with it still working from the start menu shortcuts (unless I guess those start menu shortcuts are actually pointing to a different installation - you could maybe check their 'properties' dialog to confirm they are pointed at the venv).

rpanderson, 2020-06-15:
> a) run `where blacs-gui` and `where blacs` with the venv activated and report the results and

```console
(.venv) C:\Users\rpanderson\labscript-suite>where blacs-gui
C:\Users\rpanderson\labscript-suite\.venv\Scripts\blacs-gui.exe
```

> b) confirm if `c:\users\rpanderson\labscript-suite\.venv\Scripts\blacs-gui` actually exists?

```console
(.venv) C:\Users\rpanderson\labscript-suite>dir .venv\Scripts | findstr blacs-gui
30/05/2020  04:45 PM               406 blacs-gui-script.pyw
30/05/2020  04:45 PM            75,264 blacs-gui.exe
```

Might it be the absence of the `.exe` from the relaunch entry point?

> I'm at a bit of a loss at the moment, none of the explanations I've thought of so far are consistent with it still working from the start menu shortcuts (unless I guess those start menu shortcuts [...]

The shortcut points to:
```
C:\users\rpanderson\labscript-suite\.venv\Scripts\blacs-gui.exe
```

I'll try and reproduce in another machine. This is my dev installation which has been used throughout the development of desktop-app, etc. so it might be a false alarm.

chrisjbillington, 2020-06-15:
> Might it be the absence of the .exe from the relaunch entry point?

Ah yes, it could be that. Whilst leaving out the `.exe` is fine when running in a shell, it presumably isn't fine when starting a process with `subprocess.Popen` and friends, since the shell is not involved.

If it isn't a fresh install such that you aren't running the latest `desktop-app`, then it might be [this bug](https://github.com/chrisjbillington/desktop-app/commit/e74a3c6abc9fca5f8c42bc2031ce901a143d1322) that has since been fixed.

rpanderson, 2020-06-15:
Thanks, @chrisjbillington. Upgrading `desktop-app` (and reinstalling the application shortcut for blacs) did the trick, i.e.
```
(.venv) C:\Users\rpanderson\labscript-suite>pip install -U desktop-app
Collecting desktop-app
  Downloading desktop_app-0.2.5-cp38-cp38-win_amd64.whl (23 kB)
Requirement already satisfied, skipping upgrade: importlib-metadata in c:\users\rpanderson\labscript-suite\.venv\lib\site-packages (from desktop-app) (1.6.0)
Requirement already satisfied, skipping upgrade: setuptools-scm in c:\users\rpanderson\labscript-suite\.venv\lib\site-packages (from desktop-app) (3.5.1.dev32+g8e6aa2b)
Processing c:\users\rpanderson\appdata\local\pip\cache\wheels\eb\4e\d2\a903d4184fb49e4ac06474d65715b129aee13d69f7d227e78e\distlib-0.3.0-py3-none-any.whl
Requirement already satisfied, skipping upgrade: pywin32; sys_platform == "win32" in c:\users\rpanderson\labscript-suite\.venv\lib\site-packages (from desktop-app) (227)
Requirement already satisfied, skipping upgrade: zipp>=0.5 in c:\users\rpanderson\labscript-suite\.venv\lib\site-packages (from importlib-metadata->desktop-app) (3.1.0)
Installing collected packages: distlib, desktop-app
  Attempting uninstall: desktop-app
    Found existing installation: desktop-app 0.1.3
    Uninstalling desktop-app-0.1.3:
      Successfully uninstalled desktop-app-0.1.3
Successfully installed desktop-app-0.2.5 distlib-0.3.0

(.venv) C:\Users\rpanderson\labscript-suite>desktop-app uninstall blacs
 -> deleted C:\Users\rpanderson\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\BLACS - the labscript suite.lnk

(.venv) C:\Users\rpanderson\labscript-suite>desktop-app install blacs
 -> created C:\Users\rpanderson\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\BLACS - the labscript suite.lnk
```

closed 2020-06-15 by rpanderson

chrisjbillington, 2020-06-15:
Sweet!