labscript-suite/blacs #86: Blacs fails to close when using a PluginTab

zakv, 2021-01-11:
`PluginTab` seems to be missing some methods and attributes that blacs expects tabs to have. When using a class based on the example from [here](https://bitbucket-archive.labscriptsuite.org/#!/labscript_suite/blacs/pull-requests/18/page/1), everything works as expected until blacs is closed. At that point the error below is raised and blacs fails to exit. While failing to exit it keeps logging `"INFO BLACS: destroy called"`.

```python
Traceback (most recent call last):
  File "c:\...\qtutils\invoke_in_main.py", line 46, in event
    result = event.fn(*event.args, **event.kwargs)
  File "c:\...\blacs\__main__.py", line 605, in on_save_exit
    tab.shutdown_workers()
AttributeError: 'LockMonitorTab' object has no attribute 'shutdown_workers'
```

To get blacs to exit properly I had to add the methods `shutdown_workers(self)` and `finalise_close_tab(self, *args, **kwargs)`, as well as add the attributes `state` and `shutdown_workers_complete`. Those methods don't need to do anything, but they need to be there to avoid the `AttributeError` and allow blacs to exit. Maybe the best solution is to just add these methods/attributes to `PluginTab`?

On a related note, the plugin tab doesn't have an icon and its text is black instead of blue, as expected from the link above.

chrisjbillington, 2021-01-12:
Thanks for the report. It was always silly that the PluginTab duplicates code rather than inheriting or otherwise, hence this was missed when modifying the main tab class to shutdown worker processes more cleanly.

Probably the caller should just not call the `shutdown_workers` method unless it exists.

zakv, 2021-01-13:
I briefly took a look at how hard it would be to change up the inheritance structure and it looks like it requires more knowledge of blacs internals than I have. I think your solution is pretty reasonable though and I'll probably get a chance to implement it in the next few days.

PR #87 (2021-11-24): Fixed issue where blacs couldn't close with certain tabs
Fixes #86 using Chris's suggestion there.

In short, the issue was that blacs failed to close when a `PluginTab` was used because that class doesn't have `shutdown_workers()` and `finalise_close_tab()` methods. This PR works around that by having blacs only call those methods on tabs that actually have them.
closed 2021-11-24 by dihm